FlightOne/FalcoX-Issue-Tracker #162: [FEATURE] Current Sensor Offset

Nilssonfpv, 2021-02-28:
**Is your feature request related to a problem? Please describe.**
I have two quads that both need current sensor offset to get the corect readings.
One needs -6A and the other needs -10A. Verified by a multimeter.
The builds are picking up EMI from having the Caddx vista/gps/rx cables to close.

**Describe the solution you'd like**
A configurable current offset just like in Betaflight and Inav. Just like what you have fore the voltage.

**Describe alternatives you've considered**
Iv already rebuild both quads and cleaned up the wiring as much as possible, dropping the silly amp readings from 32A in idle on both quads. to 6A & 10a at idle.

**Additional context**
Add any other context or screenshots about the feature request here.

Nilssonfpv, 2021-03-10:
> You can use the Amp Scale to adjust the The current readings, theres a also a wizard for helping set that value

I know, however there is no current offset.
And i cant find a current offset in the backup txt either, just voltage offset and current scale.

I have two quads that have unusually high idle current's. one reads 6A when motors are spinnign at idle, 10A with the caddx vista disabled.
The other reads 10A / 12A.

Besides this initial really high reading it matches with my external amp meter when i throttle up (if i factor in the 6/10A offsets manually).

PGIII, 2021-08-04:
10.1 Alpha 33 has support for a current offset now

closed 2021-08-04 by PGIII