FlightOne/FalcoX-Issue-Tracker #165: FC Diatone MAMBA F405MK2

BigVitaly, 2021-03-01:
Hello. 
FC Diatone MAMBA F405MK2 Betaflight & F40MK2 40A 3-6S DSHOT600 FPV Racing. I bought a license from you. I downloaded Windows Configurator 10.1.5-alpha. I chose the FurуF4OSD firmware. FC was stitched, and then nothing. Initially, FC Betaflight had FuryF4OSD firmware. Tell me what the problem is. 
Thank.

PGIII, 2021-03-03:
What do you mean FC was stitched ? do you have a video of what happens when you flash in DFU mode ?

BigVitaly, 2021-03-06:
Three stages of the FC firmware are carried out in the DFU mode. The configurator writes that FC will forgive. But he does not write what firmware it is stitched with. After rebooting, the FC does not connect to the configurator. The configurator does not see the port to which the FC is connected. The flight controller groans like a brick. No further adjustments are possible. After the firmware is installed in the betaflight configurator, the FC functionality is restored.

PGIII, 2021-03-17:
I’m not what you mean exactly can you show me a video of what it does when you try to flash

BigVitaly, 2021-03-20:
Hello.
I don't know how, but after updating the configurator everything worked as it should. Thank. Health. Good luck.

closed 2021-03-20 by PGIII