FlightOne/FalcoX-Issue-Tracker #169: Motors do not spin at iFlight SucceX F7 Twing BT

amusinov, 2021-03-02:
**Version**
latest available on 02 Mar 2021

**Target**
FLX_IFLIGHTF722_TWING

FC: iFlight SucceX F7 Twing BT
ESC: 4-in-1 iFlight

During motor wizard the motors do not spin at all at the stage of Idle setup (5-100%). Tried both Blheli_S and BlueJay ESC firmware (both work nice with BF at the same setup). Switching ESC settings in Configuration tab does nothing: tried multishot, Dshot etc. A series of re-flashing doesn't help. Still works on BF.

As FC rebooted, the wizard gets back to normal/reverse prop rotation screen and doesn't react on any RC commands. Only full flashing helps to restart the wizard.

Thanks.
Andrew

amusinov, 2021-03-06:
Hi guys.

Please let me know if you are planning to solve this issue soon?

PGIII, 2021-03-08:
We will look into it

jokker87, 2021-04-20:
+1 from my end, using blheli32 esc from iflight, 3 of 4 motors are spinning after the wizard when i arm the kwad, but no motors are spinning during the wizard

jokker87, 2021-04-20:
After a power cycle, the motor that doesn't spin on arming is delay with its esc/fc beeps, and you only hear 4 of the 5 beeps, so the second beep from the fc is missing. Does this have something to do with the pinout on the board, as this FC has 8 motor output pins?

closed 2021-06-10 by stale[bot]