FlightOne/FalcoX-Issue-Tracker #172: [FEATURE] Disable long beep at startup

andytroy, 2021-03-03:
**Is your feature request related to a problem? Please describe.**
My Hellgate beeper, beeps for like 8 seconds at startup. During shoots this can be a problem, because they might be recording audio and it's really annoying. I need the buzzer because I'm flying in environments where I could lose the quad.

**Describe the solution you'd like**
Make this optional or disable it, like the old firmware.

https://youtu.be/_HV2LWt4tsg

PGIII, 2021-03-08:
We can probably add an option to disable it in one of the next few alphas

kennytakfpv, 2021-07-08:
This feature would be greatly appreciated. Could you, please, add the possibility to turn off the startup buzzer beeps?

PGIII, 2021-07-09:
Startup Beep will be Disabled in the Next Alpha

PGIII, 2021-07-10:
10.1 Alpha 29 has it disabled

closed 2021-07-10 by PGIII