FlightOne/FalcoX-Issue-Tracker #174: MatekF722SE stuck on Wizard and The FC rebooting

achmiral, 2021-03-06:
**Version**
Version 10.1.0.5368 |10.1 Alpha 18|

**Target**
MatekF722SE

**Describe the bug**
Freeze on Detect VTX Wizard and The FC automatically restart

**To Reproduce**
Steps to reproduce the behavior:
1. Flash FalcoX Version 10.1.0.5368 |10.1 Alpha 18| to Matek F722SE
2. Plug in the lipo
3. Got freeze on Detect VTX Wizard

**Expected behavior**
After plug in the battery the wizard should detect the RX and the VTX

**Screenshots**
https://youtu.be/vF7LdYhbglc

**Additional context**

viktor-k99, 2021-03-08:
+1
similar bug

PGIII, 2021-03-08:
Will look into it this week

achmiral, 2021-03-08:
Great!! thank you @PGIII

viktor-k99, 2021-03-10:
Version
Version 10.1.0.5407 |10.1 Alpha 19|

Target
MatekF722SE

RC
TBS nano RX

Describe the bug
After plug in the battery the wizard not found the VTX and start search Rreciver.
The search for RX continues endlessly

To Reproduce
Steps to reproduce the behavior:

Flash FalcoX Version 10.1.0.5407 |10.1 Alpha 19|| to Matek F722SE
Plug in the lipo
Wait for the Wizard to start searching for the receiver

Expected behavior
After plug in the battery the wizard should detect the RX and the VTX

viktor-k99, 2021-03-10:
@achmiral  what version of TBS nano RX firmware do you have?

achmiral, 2021-03-10:
@viktor-k99 3.78

viktor-k99, 2021-03-11:
@achmiral I will downgrade the firmware in the receiver and try again

viktor-k99, 2021-03-13:
I downgraded the Receiver firmware to 3.78, tried to connect  Reciver to UART1 or UART2. The wizard didn't detect the receiver.

I have flashed the Receiver firmware to version 4.10 and soldered the receiver to UART1. I flashed the FC into Betaflight and checked that Betaflight detects the receiver correctly.

I flashed the FC firmware to FalcoX Alpha Version 10.1.0.5414 | 10.1 Alpha 20 |. Updated configurator to version 10.1.6-alpha.
The wizard does't detect the Reciver on UART1.

PGIII, 2021-04-13:
This crashing was fixed in 10.1 Alpha 19

closed 2021-04-13 by PGIII