FlightOne/FalcoX-Issue-Tracker #175: Yaw twitches when mixing Yaw and Roll on Lightning H7

Georgeandrew7, 2021-03-07:
**Version**
10.10.5368

**Target**
Lightning H7 Wild Willy 

**Describe the bug**
Mixing yaw and roll together causes yaw twitches and yaw spins of death. 

**To Reproduce**

1. Attempt to fly 
2. Mix yaw and roll together going in the same direction. 
3. Try both directions 

**Expected behavior**
Reliable flight characteristics. 

Tried all tuning presets.

PGIII, 2021-03-07:
That sounds like an ESC desync, what are your rates set to and what esc and esc settings are you using

Georgeandrew7, 2021-03-07:
Esc settings: demag compensation high,
Timing 25 deg, frequency 24khz

Georgeandrew7, 2021-03-07:
Rates: Vanover v2 preset

PGIII, 2021-03-08:
I would try 32 or 48khz for Motor PWM

Georgeandrew7, 2021-03-08:
Okay, thank you! I'll keep you posted on the results.

closed 2021-06-10 by stale[bot]