FlightOne/FalcoX-Issue-Tracker #176: Missing VTX MW on OSD element

achmiral, 2021-03-09:
**Version**
Version 10.1.0.5407

**Target**
Currently tested on MatekF722SE and HGLRC F722

**Describe the bug**
VTX MW is not available on osd element list

**To Reproduce**
Steps to reproduce the behavior:
1. Go to OSD menu
2. On OSD page, select setup osd
3. No "VTX mw" on the list
4. See error

**Expected behavior**
There is "VTX mw" element on osd element list

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

Rafyel, 2021-03-13:
PLEASE FIX THIS ASAP!!!!

PGIII, 2021-03-13:
This should be fixed in the next alpha

PGIII, 2021-03-13:
> PLEASE FIX THIS ASAP!!!!

This is an alpha version that’s not even been out for a week yet and you can always see Power level in the menu, so there’s no need to freak out it should be fixed in the next alpha

achmiral, 2021-03-13:
Great!!, thanks @PGIII 👍

PGIII, 2021-04-13:
Has been fixed in previous updates

closed 2021-04-13 by PGIII