FlightOne/FalcoX-Issue-Tracker #177: Motor 2 Not Spinning on MatekF722SE

achmiral, 2021-03-10:
**Version**
Version 10.1.0.5407 |10.1 Alpha 19|

**Target**
MatekF722SE

**Describe the bug**
Motor 2 not spinning, even when I reconfigured the wizard several times.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash FalcoX 10.1.0.5407 to MatekF722SE
2. Plug the battery and start the wizard
3. Oh the motor idle wizard, the motor 2 is not spinning
4. See error

**Expected behavior**
All the 4 motors spinning, I've tried to flash the FC with betaflight, all motors are spinning as the I expected.

**Screenshots**
https://youtu.be/8F37RFx5NxU

WilidWilly, 2021-03-13:
[FLX_MATEKF722SE_10.1.0.5417.zip](https://github.com/FlightOne/FalcoX-Issue-Tracker/files/6134302/FLX_MATEKF722SE_10.1.0.5417.zip)
It appears the pin timer wasnt mapped right, give this fw a shot and it should work fine now

achmiral, 2021-03-13:
Thank you @WilidWilly 👍

WilidWilly, 2021-03-14:
@achmiral did that bin work?

achmiral, 2021-03-14:
@WilidWilly yup, it is work, thank you 👍

closed 2021-03-15 by PGIII