FlightOne/FalcoX-Issue-Tracker #179: 10.1.6 Configurator can't find CRSF remote on iFlight TWING F722

marcelinojorgeromero, 2021-03-12:
**Describe the bug**
I updated the configurator to the latest version (10.1.6) and then updated the firmware (and bootloader) of my TWING F722 quad then i turn on the radio and the configurator couldn't find the radio. I know for sure the radio is communicating with the receiver because all of the indicators i have (and it was working before the update), just the configurator can't find it.

**To Reproduce**
Steps to reproduce the behavior:
1. Updated the configurator and firmware to the latest (Configurator 10.1.6), firmware 10.1.0.5414
2. Waiting for the configurator to recognize the 

**Expected behavior**
I was expecting the configurator to connect to my radio.

**Screenshots**
[image]

Diagram
[image]

**Desktop (please complete the following information):**
 - Windows 10
 - Chrome 
 - Version 88.0.4324.190

**Additional context**
I also run _resetconfig_ but had same results.

PGIII, 2021-03-12:
What version was on the board before ? And is the light blinking normally ? Also configurator doesn’t have any effect on the Rx detection so sounds like a fw related issue

marcelinojorgeromero, 2021-03-13:
> What version was on the board before ? And is the light blinking normally ? Also configurator doesn’t have any effect on the Rx detection so sounds like a fw related issue

I think it was on alpha 15. There are a red light (not blinking), a blue light blinking and a green light from the crossfire module. What if the new telemetry function broke the connection? I haven't updated my opentx and crossfire firmware for a while.

PGIII, 2021-03-15:
The telem just adds support for new frames and responds to them, don't see how it could really make crossfire not work, did you reset the config after you upgraded ? if you had it detected before it shouldn't have to do the detection again

Naimh, 2021-03-15:
Exactly have the same issue with the same hardware (wich was working well under BF), CRSF is not detected

marcelinojorgeromero, 2021-03-20:
still does the same on the latest firmware update.
1 red led solid light, 1 blue blinking (looks like 1 time per second) and the green solid slight from crossfire but the configurator still runs through the loop without finding the crossfire connection (firmware v3.78).
[image]

PGIII, 2021-04-13:
Are you sure it detected fine on a older version and if so what was that version ? also have you tried setting the crossfire to output to SBUS to see if that detects

marcelinojorgeromero, 2021-04-13:
> Are you sure it detected fine on a older version and if so what was that version ? also have you tried setting the crossfire to output to SBUS to see if that detects

Yes absolutely sure. In fact on that quad i've created the issue #133 (Acc Read) before. I didn't change anything afterwards, just updated the firmware.
How do I change the output to SBUS? Is this a configuration trough the micro tx or should i change the cables? The FC is F7.

PGIII, 2021-04-20:
https://shop.iflight-rc.com/index.php?route=product/product&path=20_27_134&product_id=993 seems to be detecting fine with this FC on that target for me, but do you know what Gyro the board youre using has on it ? Could be some setting wrong with the target but not sure why the board i have here would work fine in that case

jokker87, 2021-04-20:
My https://shop.iflight-rc.com/index.php?route=product/product&product_id=1043 target finds CRSF flawlessly on latest alpha

marcelinojorgeromero, 2021-04-20:
> ould work fine in that case

how could i find the gyro version? Is there any command i could use to dump all the data?

jokker87, 2021-04-21:
@marcelinojorgeromero did you try to connect battery first and then usb cable? i had issues with not dedecting CRSF when doing it the other way around. i'm using the latest CRSF version, so maybe updating the CRSF TX + RX could help.

marcelinojorgeromero, 2021-04-21:
> @marcelinojorgeromero did you try to connect battery first and then usb cable? i had issues with not dedecting CRSF when doing it the other way around. i'm using the latest CRSF version, so maybe updating the CRSF TX + RX could help.

yes i tried all possible combinations and resetting multiple times and i uninstalled the software once. I tried with battery on and off then turned on and off the radio etc.

PGIII, 2021-04-21:
Could be something to do with CRSF version, ive tested on 4.1 and 4.2 recently, might be worth trying 4.1 and see if that changes anything the TBS agent wont even let me downgrade to 3.78 it seems, as far as i can tell 4.1 is the latest stable now

ocservo, 2021-05-31:
I had the same situation and I also experimented with DJI's air unit.So far I haven't solved the problem

crashingdrones, 2021-06-17:
this is still broken, flight twing v2.1 f7 hd.... doesn't detect remote, using dji remote....

PGIII, 2021-08-14:
[GYRO_FIX_IFLIGHTF722_TWING_10.1.1.5703.flx.zip](https://github.com/FlightOne/FalcoX-Issue-Tracker/files/6986774/GYRO_FIX_IFLIGHTF722_TWING_10.1.1.5703.flx.zip)
Try this and see if it fixes it

ocservo, 2021-08-15:
Let me try