FlightOne/FalcoX-Issue-Tracker #180: [FEATURE] ExpressLRS doesnt support on furyF4OSD

ridwannurfalah, 2021-03-13:
r9m mini receiver using expresslrs was not detected.

PGIII, 2021-03-15:
What protocol are you using ? and how do you have it connected

ridwannurfalah, 2021-03-15:
i’m using EXpressLRS flashing via betaflightpassthrough method, everything going successfull flashing…. and connect (RX) via A1 to tx1 (FC) , A2 to RX (fc), gnd and (+)… but on Falcox alpha cannot find radio rx (still searching)until 10 minute

ridwannurfalah, 2021-03-15:
> What protocol are you using ? and how do you have it connected

using crsf on tx... and connected the pin a1 &a2 on rx to FC

PGIII, 2021-03-16:
so the protocol the receiver is outputting is CRSF ?

ridwannurfalah, 2021-03-17:
> so the protocol the receiver is outputting is CRSF ?

flashing instructions r9mini / mm using CRSF https://github.com/ExpressLRS/ExpressLRS/wiki/OpenTX-and-Betaflight-Setup

AlessandroAU, 2021-03-18:
Is serial passthrough supported on F1? @ridwannurfalah log of the console output when flashing is attempted would be helpful.

ridwannurfalah, 2021-03-29:
> Is serial passthrough supported on F1? @ridwannurfalah log of the console output when flashing is attempted would be helpful.

this notification from log file, : data: '*** [upload] Exception : No valid serial port detected or port already open\n'
  },
  level: 'info',
  message: 'logs stream output'

PGIII, 2021-04-01:
Theres no UART to USB Serial passthrough, are you trying to flash the Rx or are you having issues detecting ?

closed 2021-06-10 by stale[bot]