FlightOne/FalcoX-Issue-Tracker #181: H7 problem with blheli32 passthrough ESC4 flashing always fails

zeezyfpv, 2021-03-14:
**Version**
Latest alpha 10.1.0.5414 or stable 10.0.18.4939

**Target**
H7 Lightning wildwilly edition

**Describe the bug**
esc4 flash fail (blheli32)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

On my 2 different H7 setups got ESC4 flash fail in blhelisuite32. Escs are aikons ak32 and ak32pin 4in1 35A. Tried different blhely32suit versions,esc firmware versions,H7 with stable or latest firmware 10.1 alpha 20. Suite reads and writes esc settings fine but esc4 flash always fails. Unhooked everything from H7,just cable for escs. When I hook betaflight fc to escs everything works fine,no esc4 flash fail.

PGIII, 2021-03-15:
What version of Blheli suite and blheli32 are you using

zeezyfpv, 2021-03-15:
Tried different blhelisuite versions..32.8 beta, 32.72, 32.70 and aikon hex files 32.6, 32.7 and 32.74 with var.pwm. Doesn't matter,with H7 esc4 flash always fails and won't reflash successfully until I hook betaflight fc on aikons...in my case

PGIII, 2021-03-16:
What version did you try with the current alpha ? there has been changes related to passthrough between the stable and alpha, the only issue I know of with the current alpha was when using the 32.8 blhelisuite

PGIII, 2021-04-13:
Closing Due to no activity and has been reported to be fixed in latest version

closed 2021-04-13 by PGIII