FlightOne/FalcoX-Issue-Tracker #183: |10.1 Alpha 20| with H7

David180680, 2021-03-15:
GPS Wizard Highest point:
there is a bug when i have to fly 5s , my quad turns on itself when I put the gas on

PGIII, 2021-03-15:
Im not really sure what you mean and are you saying you're having this issue during the GPS wizard?

David180680, 2021-03-16:
Yes I'm having this issue on the GPS Wizard

PGIII, 2021-03-16:
But what exactly is happening your description isn't super clear, did you happen to get a DVR of the issue ?

David180680, 2021-03-16:
see dvr on the comment of Preston Garrison : https://www.facebook.com/groups/flightone/permalink/4110452725673165/

David180680, 2021-03-22:
problem solved by following advice from Wild Willy : " I think you may have to arm and hover the quad first though before running the RTH setup wizard to get the gyro and accelerometer data running right"

closed 2021-06-10 by stale[bot]