FlightOne/FalcoX-Issue-Tracker #186: Not able to change pilot name from FlightOne

Georgeandrew7, 2021-03-15:
**Describe the bug**
When attempting to save any other pilot name except for FlightOne, I get error saving code: 30002

Version: 10.1.0.5414 Alpha 20

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the OSD menu
2. Set wanted pilot name 
3. Save 
4. See error 30002

**Expected behavior**
Saving of pilot name and seeing it in the OSD.

PGIII, 2021-03-16:
Will see if I can replicate it, what did you set the pilot name to ?

Georgeandrew7, 2021-03-16:
I tried every given name I could think of, even the name of the quad. Nothing but FlightOne would save properly. I believe I tried twenty or so different names and character combinations to see if it would save anything but FlightOne. The most prominent one I attempted was Switchback1.

2dogrc, 2021-03-22:
@PGIII This also does not work in the configurator on the PC.  Have to use the Terminal.

closed 2021-04-19 by PGIII