FlightOne/FalcoX-Issue-Tracker #189: [FEATURE] Toggle air mode on switch

Naomarik, 2021-03-23:
**Is your feature request related to a problem? Please describe.**
When coming into land gently at 0 throttle and touching the ground, you the quad likes to bounce. Just like Betaflight's air mode.

This makes perching on objects in Falco X impossible and skidding on the ground. It also requires you to be extremely accurate on disarming, otherwise the quad bounces or you hit the ground hard.

**Describe the solution you'd like**
Allow assigning a switch to toggle  the "air mode" effect, putting drone into exactly what Betaflight's "acro" mode is.

**Additional context**
This makes it possible to pull moves typically only seen by KISS pilots.

PGIII, 2021-03-23:
Its not something that can just be turned off, we give you control over the quad 100% of the time, you can tune down the TPA at lower throttles to reduce how much it reacts when you hit things, in general Bouncing When you hit the ground is usually due to too high of PIDs, More filtering would also help also

Naomarik, 2021-03-23:
I can try that, but wouldn't that affect 0 throttle flips? Bouncing is normal in Betaflight when air mode is regardless of PIDs. I've always assigned it on a switch. 

I've also had situations when filming where I needed my quad to be somewhere away from me but not disarmed. Was able to just drop it down while keeping motors spooled and take off at a moments notice.

Naomarik, 2021-03-23:
[image]

If there was just one more notch in this screen, say at like ~3%, I could set the PIDs low at 0% throttle, then 100% at 3%. Then could I could use a switch to change the idle speed on my transmitter.