FlightOne/FalcoX-Issue-Tracker #190: Matek F722-SE doesn't find TBS crossfire receiver

viktor-k99, 2021-03-24:
**Version**
Version 10.1.0.5407 |10.1 Alpha 19|
....
Version 10.1.0.5447 |10.1 Alpha 23|

**Target**
MatekF722SE

**RC**
TBS nano RX version 3.78
TBS nano RX version 4.10
TBS nano RX version 4.10

**Describe the bug**
After plug in the battery the wizard not found the VTX and start search Reciver.
The search for RX continues endlessly

**To Reproduce**
**Steps to reproduce the behavior:**
Solded TBS nanorRX (rev. XFnano 45) on UART2 (or UART1, UART3, UART4)
Flash FalcoX Version 10.1.0.5447 |10.1 Alpha 23| to Matek F722SE
Plug in the lipo
Wait for the Wizard to start searching for the RC receiver.

**Expected behavior**
After plug in the battery the wizard should detect the RX.

**Screenshots**
https://youtu.be/OU81sKkArKo

dieselshroom, 2021-03-25:
Just had this exact same issue when trying to run the wizard on this board. Had no choice but to go back to BF.

PGIII, 2021-04-13:
What Speed is the crossfire set to and does it detect if you set the crossfire to ni sbus ?

viktor-k99, 2021-04-13:
My settings Crossfire Micro TX
Multibind -    Enable
Region -        Open
Max Power - 100mW
Dyn.Power -  On
Frequency -  868 Mhz
Op. Mode  -  Normal

Settings Nano RX
Mode - 8 Ch
Telemetry - On
RF Profile - Dynamic
Failsafe Mode - Cut
RC by MAVLink - oh
RX Batt Sensor - OFF
RX Output map:
Output 1 - CRSF TX
Output 2 - CRSF RX

I set the following options:
- RF Profile - 150 Hz, Mode 8 ch
- RF Profile - 150 Hz, Mode 12 ch
- RF Profile - 50 Hz, Mode 8 ch
- RF Profile - 50 Hz, Mode 12 ch
- RF Profile - Dynamic, Mode 8 ch
- RF Profile - Dynamic, Mode 12 ch
 The wizard doesn't  detect the RC receiver Crossfire nano RX.
I set Output 1  at n.i.SBUS. The wizard didn't  detect the RC receiver Crossfire nano RX.
I reset my Crossfire Micro Tx and Nanor Rx to factory settings . The wizard doesn't  detect the RC receiver Crossfire nano RX again.

yyacoel, 2021-04-29:
Same issue for me. with xfire nano.
Tried to use different UARTs (1/2/3), same behaviour.

viktor-k99, 2021-04-29:
I think that the support team needs to test the firmware on the Matek board. Only after tests can the this issue be resolved.

ojwright845, 2021-06-02:
Same issue with matek f722se. Frsky receiver not detected. Attempted manual configuration but no luck

GFRaw, 2021-06-03:
Also stuck with unable to detect the RX, crsf nano.  Going back to BF for now.

3D-pilot, 2021-06-09:
As per viktor-k99 Matek F722SE Unable to detect TBS Nano Rx. Configurator keeps continuously cycling through RX options and unable to detect. Tried RESETCONFIG and various Uarts.

PGIII, 2021-08-14:
[GYRO_FIX_IFLIGHTF722_TWING_10.1.1.5703.flx.zip](https://github.com/FlightOne/FalcoX-Issue-Tracker/files/6986810/GYRO_FIX_IFLIGHTF722_TWING_10.1.1.5703.flx.zip)
try this and see if it fixes it

closed 2021-09-20 by stale[bot]