FlightOne/FalcoX-Issue-Tracker #193: Crossfire LQ (RSSI) on DJI OSD showing values from 1 to 9 instead of 0 to 100%

felipeng, 2021-03-28:
**Version**
What version are you using:
* FalcoX: 10.0.18.4939
* Vista: V01.00.0600
* Goggles: V01.00.0606

**Target**
What target are you using: RevoltOSD

**Describe the bug**
Crossfire LQ on DJI OSD shows values from 1 to 9 instead of 0 to 100%

**To Reproduce**
Steps to reproduce the behavior:
1. Power on kwad/goggles/radio

**Expected behavior**
Range values from 0 to 100%

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
* Connecting the kwad on the USB in the FalcoX Configurator LQ values shows correctly from 0 to 100%
* Would be awesome to set up an alert on low LQ, like 40% ow lower set the OSD element in red

PGIII, 2021-04-01:
Will have to check probably a scaling issue, alerts are up to DJI

AndroidGX, 2021-04-21:
Same issue here I confirm.
My LQ is set on channel 8 and selected RSSI into OSD but its stuck in "9" value once battery plugged.

famascl3m, 2021-05-12:
Same issue using crsfr telemetry directly through the configurator, value is good in the configurator OSD but not in the googles OSD

felipeng, 2021-06-11:
Same issue with latest version