FlightOne/FalcoX-Issue-Tracker #200: New ClracingF7 AIO V2 target , no video output from FC

Peryshev, 2021-03-29:
Version
Version 10.1.0.5447 |10.1 Alpha 23

Target
CLracing F7

Describe the bug
New ClracingF7 AIO V2 target , no video output from FC . I can see black screen only .

In Betaflight  Set mco2_on_pc9=on make video work , this command was added to Apply Custom Defaults for V2 boards. On V1 board video works with mco2_on_pc9=off. Not sure but mco2_on_pc9=on was added to V2 board for DJI video system .

I tested  "set mco2_on_pc9=off"   in Betaflight turn video to black screen same what I can see on FalcoX!

PGIII, 2021-04-01:
There's no support for MCO right now, Not sure when we will add that ill need to get a Board that uses it to test here, that being said you said you said its a DJI version ? are you using DJI or analog video ?

Peryshev, 2021-04-01:
> There's no support for MCO right now, Not sure when we will add that ill need to get a Board that uses it to test here, that being said you said you said its a DJI version ? are you using DJI or analog video ?

HI PGIII Im guessing that DJI support was one of the reason why Clracing change that think when they make V2 board. Am using it with analog.

Peryshev, 2021-04-04:
> There's no support for MCO right now, Not sure when we will add that ill need to get a Board that uses it to test here, that being said you said you said its a DJI version ? are you using DJI or analog video ?

I text you in FB pleas check my massage. I need your address to send you a board for testing !