FlightOne/FalcoX-Issue-Tracker #202: Spinning uncontrollably after crash on H7

Georgeandrew7, 2021-04-01:
**Version**
10.1.0.5467 Alpha 24 

**Target**
Lightning H7 WildWilly 

**Describe the bug**
Crashing into a gate, the quad spins uncontrollably and unable to recover with the spin fix. It keeps spinning in one direction, then the other, and eventually hits the ground unless disarming and rearming quickly to recover. 

**To Reproduce**
Steps to reproduce the behavior:
1. Fly 
2. Crash and see if it spins uncontrollably 

**Expected behavior**
Quad flights any external force, not causing it to spin uncontrollably.

PGIII, 2021-04-01:
you can increase SpinFix Recovery to give the ESC more time to recover, is it showing that spin fix is activating ?

Georgeandrew7, 2021-04-01:
Increased SpinFix recovery, it did show that it was activating in the status, but never fully recovers until I disarm and rearm the quad.

PGIII, 2021-04-05:
What do you have it set to now ? you can increase it more also i would make sure the ESCs are set to demag high, do you have any DVR of it ?

Georgeandrew7, 2021-04-05:
I have it set to 220ms and 170ms. I also have demag set to high, but still seems to occur frequently. I will get some DVR footage of it when able.

PGIII, 2021-04-13:
likely just need to increase recovery time some more, you can also adjust the other settings to make it activate sooner, what motors and escs are you using ?

Georgeandrew7, 2021-04-13:
I've increased it to 190 and 270. Spin fix now no longer activates even after resetting the config. The motors I am using are the Five33 Headsup 1960kv motors and the ESC is a Hobbywing Xrotor 40A 4in1 Blheli32 ESC.

PGIII, 2021-04-13:
What number are you increasing ? recovery affects how long it stays active for,  SpinFix Time To Enable changes how long it has to see high error before it enables Spin Fix, and SpinFix Max Error sets the threshold for what high error is. So if youre having issues with it activating you should decrease SpinFix Max Error or SpinFix Time To Enable

jokker87, 2021-04-24:
Same problem on a HGLRCF722 target. Why is this even needed? BF just recovers from a crash.

closed 2021-06-10 by stale[bot]