FlightOne/FalcoX-Issue-Tracker #203: Failsafe not working on ExpressLRS

keczejo, 2021-04-05:
**Version**
10.1.0.5470

**Target**
RevoltOSD
**Describe the bug**
Drone don't disarm when signal is lost, but when signal comes back drone disarms and shows FAILSAFE in arming status.

**To Reproduce**
Steps to reproduce the behavior:
1. Arm drone
2. Turn off radio controller or remove TX module
3. Turn radio back on or reinstall TX module

**Expected behavior**
Drone should disarm when signal is lost.

**Viedo with problem**
https://drive.google.com/file/d/10t0LC5ZMzBdyH2GwVjEoxJGjKApP3_Vi/view?usp=sharing

PGIII, 2021-04-05:
Looks like the Rx is still sending Packets, what protocol does it use ? also im not familar with expressLRS but how do you have the  failsafe setup?

keczejo, 2021-04-05:
> Looks like the Rx is still sending Packets, what protocol does it use ? also im not familar with expressLRS but how do you have the  failsafe setup?
> 

Expresslrs uses crsf protocol. Elrs have no failsafe options, but failsafe on betaflight is detected properly.

NelisFPV, 2021-04-06:
experiencing the same issue using tracer RX with crsf protocol.

PGIII, 2021-04-10:
[FLX_REVOLT_10.1.0.5471.zip](https://github.com/FlightOne/FalcoX-Issue-Tracker/files/6290645/FLX_REVOLT_10.1.0.5471.zip)
 try this one

PGIII, 2021-04-12:
The changes fixed #204 so it likely should fix this issue

keczejo, 2021-04-12:
It fixed the problem

closed 2021-04-12 by keczejo