FlightOne/FalcoX-Issue-Tracker #204: failsafe doesn't work. motors keep spinning

NelisFPV, 2021-04-06:
**Version**

10.1.0.5470 

**Target**

foxeerf722v2

**Describe the bug**

Quad will not go into failsafe (motors cut) when turning off the transmitter. motors keep spinning.

**To Reproduce**

- arm quad
- throttle up 20-30%
- turn off transmitter

**Expected behavior**
motors turn off within a reasonable time (1s?) after turning off the transmitter.

**Additional context**
using CRSF protocol on tracer RX. tango2 transmitter.
failsafe on betaflight worked with this same setup.
RX failsafe is set to CUT.

sniperxnl, 2021-04-06:
I have the same issue ony my quad with CRSF and Tracer.
RX is set to CUT as failsave
FC Foxeer F722v2

PGIII, 2021-04-07:
Can you get a DVR of it with the RxFPS on the screen

NelisFPV, 2021-04-07:
Will do man, will send you tonight (european time)

NelisFPV, 2021-04-07:
@PGIII dvr with RxFPS in osd. also rssi showing.

https://www.dropbox.com/s/wzl8b22ci4akgem/MOV210407-202714-000425F.MOV?dl=0

PGIII, 2021-04-10:
[FLX_FOXEERF7V2_10.1.0.5471.zip](https://github.com/FlightOne/FalcoX-Issue-Tracker/files/6290642/FLX_FOXEERF7V2_10.1.0.5471.zip)
Try this and see if it fixes it

NelisFPV, 2021-04-10:
Thanks, will give it a try tomorrow and send you an update here.

NelisFPV, 2021-04-11:
@PGIII gave it a try and the failsafe works again! status failsafe is showing in osd and motors turn off within one second.
great work!

PGIII, 2021-04-12:
Sounds good next alpha should have this fix included

closed 2021-04-12 by PGIII