FlightOne/FalcoX-Issue-Tracker #205: Airbot F7 HDV stuck in wizard

diankurus, 2021-04-08:
**Version**
Version 10.1.0.5467

**Target**
AirbotF7HDV

**Describe the bug**
Cant pass the fc direction setup. When quad is laying flat and roll left. It said ACC Z 0

**Expected behavior**
It passed the wizard

**Screenshots**
[image]

PGIII, 2021-04-12:
Dont think anyones tested this target yet, does it show the right target when it connects that screen shot is pretty low res its hard to tell

diankurus, 2021-04-14:
[image]

Here the higher resolution. Error on wizard

closed 2021-06-10 by stale[bot]