FlightOne/FalcoX-Issue-Tracker #206: Strange dnoise, mtos, on alpha 24 and above

d00m303, 2021-04-08:
**Version**
alpha24 and above

**Target**
H7 Lighning

**Describe the bug**
Newer alpha introduced vibrations, hot motor and bad sound motors. Tested on some bins
Alpha 15 - smoothest, cold motors
Alpha 23 - still cold motors, little more mtos than 15 but ok
Alpha 24 and above - sounds bad and flies worse than before
U can hear typical dnoise at about 30% thr, its untunable, u cant filter it.
I tested on switching bins - alpha15 cleanest
Setup was same, just i was switching bins thru disk on H7, config is restored from saved file.
There is something between 23 and 24 that causing problems, its pretty much causing noise and bad hd.

**To Reproduce**
Steps to reproduce the behavior:
1. Put T% on osd
2. Fly straight line without touhing controller at 25-35% throtlle
3. Listen the sound of motors
4. Check for mtos

**Expected behavior**
Clean hd picture and good motor sound

**Screenshots**
https://www.youtube.com/watch?v=zPaxpStYepg alpha24 - characteristic noise sound and really shaky footage
https://www.youtube.com/watch?v=5o0vZ8-h2wI alpha23 - clean motors sound, lot less mtos, pretty good
https://drive.google.com/file/d/1I0LPTvf2hKGNPbjsbcMIHqw5fZaJbQv-/view?usp=sharing alpha15 - great hd picture (sorry for no sound, i deleted file from sd ;/)

**Additional context**
Please fix this, do the mid throtlle from alpha15 with great propwash handling like alpha23, thank You! ;)

PGIII, 2021-04-09:
[Test FW Download](https://drive.google.com/open?id=19EQb6PsyibRP7UvzAqOjsTwxhTME-WpJ&authuser=pg3%40flightone.com&usp=drive_fs) 
Try 10.1.05450 and 10.1.0.5457

jadirc, 2021-04-12:
It seems that I have a similar problem. The motors sound like they are constantly stuttering. However, I have a new AOI board with a MatekF411 as target and BLHeli_s ESCs. I use dshot8 for this (others like dshot600 do not work at all!). Tested with ESC firmware JESC 24/48kHz and BlueJay 48kHz. With Betaflight and Emuflight I don't have these problems. I would like to test an older FX firmware, but unfortunately there is no history of the old firmware on the website. @PGIII  can you possibly provide me an alternative version for testing? Thx

d00m303, 2021-04-12:
5450 - a lot better than alpha 25, motors sound cleaner for sure.
But alpha15 still better HD a tiny bit, smoother

PGIII, 2021-04-12:
> 5450 - a lot better than alpha 25, motors sound cleaner for sure.
> But alpha15 still better HD a tiny bit, smoother
Did you try 5457 i need to know when it starts having the issue

PGIII, 2021-04-12:
> It seems that I have a similar problem with version 10.1.0.5470. The motors sound like they are constantly stuttering. However, I have a new AOI board with a MatekF411 as target and BLHeli_s ESCs. I use dshot8 for this (others like dshot600 do not work at all!). Tested with ESC firmware JESC 24/48kHz and BlueJay 48kHz. With Betaflight and Emuflight I don't have these problems. I would like to test an older FX firmware, but unfortunately there is no history of the old firmware on the website. @PGIII can you possibly provide me an alternative version for testing? Thx
Dont know that is related to this, the MatekF411 has been support for a while there should be a beta version for it

d00m303, 2021-04-12:
> > 5450 - a lot better than alpha 25, motors sound cleaner for sure.
> > But alpha15 still better HD a tiny bit, smoother
> 
> Did you try 5457 i need to know when it starts having the issue

didnt tried, ill check it next, i had hope u knew what it can be so i flashed latest which u posted here ;D

PGIII, 2021-04-30:
Reported as being fixed in 10.1 Alpha 27

closed 2021-04-30 by PGIII