FlightOne/FalcoX-Issue-Tracker #210: The motors sound like they are constantly stuttering

jadirc, 2021-04-14:
**Version**
10.1 Alpha 26

**Target**
MatekF411and JHEF411

**Describe the bug**
I have 2 identical copters (motors, frame) with 2 different AOI FCs (GHEF411 and MatekF411). I have flashed the latest alpha version on both. As soon as I give a little more throttle the motors start to stutter continuously. For testing I flashed one with the version 10.0.6.3923 (MatekF411) and then the problem did not occur.
Tested with ESC firmware JESC 24/48kHz and BlueJay 48kHz. With Betaflight and Emuflight I don't have these problems too.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash the newest Alpha Version with given Target
2. Use Dshot8
3. Arm Motors
4. Hear the sound of the motors

notomatoplease, 2021-04-16:
I have the same issue on Alpha firmware with a betafpv toothpick fc (MatekF411). Dshot600 with JESC works decently but doesn't completely get rid of the stuttering. Bluejay esc firmware only works on Dshot8 and stutters a lot. I tried different presets and they all had this issue.

PGIII, 2021-04-20:
Can you get a Video of the issue? also have you tried the Stable version and were you using dshot8 on the beta version also ?

jadirc, 2021-04-21:
@PGIII I have only tested the current beta version for the MatekF411 target and Dshot8 so far. As I wrote above, the problem no longer occurs with it. I have not tested the stable version. I will try to upload a video in the next few days where you can "hear" the problem. 
I suspect that this problem was added in one of the latest alpha versions. I can also try older alpha versions if you can provide me with them.

notomatoplease, 2021-04-30:
https://user-images.githubusercontent.com/82621122/116643678-73b49d00-a926-11eb-9786-014a8e6f9bf5.mov

This is Alpha 5490 on Dshot8. Dshot600 is pretty good but still has an occasional stutter. Stable on Dshot8 is like Alpha on Dshot600, occasional stutters. Beta 3923 on Dshot8 is smooth without any issues.

PGIII, 2021-04-30:
Can you flash to Alpha 27 and go into system info and get a DVR of the Serial Info page

notomatoplease, 2021-04-30:
https://user-images.githubusercontent.com/82621122/116759699-d742d700-a9c7-11eb-9739-d73beed7e2a3.mov

This?

closed 2021-06-10 by stale[bot]