FlightOne/FalcoX-Issue-Tracker #211: [FEATURE] Shark Byte Integration

steeevefpv, 2021-04-15:
As FS released the new firmware for sharkbyte system featuring the "canvas mode" it should now be easier to interact with other flight controller/firmware.
They suggest to update the Displayport code of the fc.
This is from one of their last update mail
"There have been a lot of requests for KISS support (and other flight controllers). We have been working hard on initially supporting Beta Flight and being sure all of the features are there with a lot of recent work on Display Port code that we wanted to get out. We are now in contact with developers for other flight controllers and suggest that everyone doing development update your Display Port code so that the next f/w update of Byte Frost with canvas mode will work. "

And this is from the maunal:
"CANVAS MODE (BETAFLIGHT)
Currently only Betaflight supports msp_displayport. Do not attempt to perform this CLI update on INAV. Please make sure that the FC firmware version is the latest
version of BetaFlight. The current (2021/03/12) is version 4.2.8. For a better experience, please wait for the release of version 4.3.0.

Connect VTX UART to FC’s serial port, such as UART3.
In the Ports interface, turn on the MSP function corresponding to the Identifier (UART3), the baud rate is 115200, Save and Reboot.
After opening the CLI, make the following settings:
A. Set osd_device to msp:
set osd_displayport_device = MSP
B. Specify the serial port of msp_displayport as 2 (the number in this place should be the serial port number minus 1):
set displayport_msp_serial = 2
C. Save and exit:
save
The setup is complete"
Please please please, work on it!

ojwright845, 2021-06-02:
Any update on sharkbyte integration?

FastJack2, 2021-07-05:
Any news on this? would love to convert my whole fleet to FalcoX but havong no OSD is an absolute killer on this

closed 2021-08-03 by PGIII

Tigereye2k6, 2021-10-30:
The OSD stopped working with the latest SharkByte Firmware 20102021. I have flashed the latest FalcoX Firmware but it still wont work. With the previous SharkByte Firmware version all worked well.

The max buad rate is set correct to 115200 and i see sometimes the start message "FALCOX OVER DISPLAY PORT" but this message stays and the osd wont get displayed. Most of the times nothing will be displayed.

Then i have flashed BetaFlight and there the OSD works as expected without any issues :/

Tigereye2k6, 2021-10-30:
Here is the new Issue:

https://github.com/FlightOne/FalcoX-Issue-Tracker/issues/279