FlightOne/FalcoX-Issue-Tracker #212: Rush Tank Solo vtx power level issue

steeevefpv, 2021-04-16:
**Version**
Version 10.1.0.5480 |10.1 Alpha 26|

**Target**
Lightning H7 WW edition

Just added a Rushfpv tank solo (Rush newest vtx), I'm on latest alpha, it seems that power level aren't working well. 
Tried every preset but still can't get the right level: 25mw is ok, at 100/200 (depending on the preset) i still get 25mw , at 400/500 i get the 400, at 800/1000 I get 800 or goes back to 25, cannot get the maximum output.
From the manual the vtx power levels should be: PIT/25/400/800/max power (1000+mW)
Any idea?

PGIII, 2021-04-19:
If it uses Smart audio the power is simply different levels, so just need to pick a model that uses the same version of SA and has the same number of power levels

steeevefpv, 2021-04-20:
I've tried every model but no success. As far as I know smartaudio 2.1 uses dbm values and following BF vtx table the values for rush tank solo should be 
"powerlevels_list": [
            {
                "value": 14,
                "label": "25 "
            },
            {
                "value": 26,
                "label": "400"
            },
            {
                "value": 29,
                "label": "800"
            },
            {
                "value": 32,
                "label": "MAX"
            }
        ]

Maybe this vtx need a separate model?

PGIII, 2021-04-20:
We use the power levels for all SA VTX's right now, We only directly set the dBm value to get it to 0mw i believe. But would be weird if they didnt map all the values to power levels thats pretty standard for TBS. In fact i think the TBS Devices round up to the power levels when you set them directly. We could probably change the way SA2.1 sets the power levels but this is the only VTX ive heard of having to be set that way

steeevefpv, 2021-04-20:
> We use the power levels for all SA VTX's right now, We only directly set the dBm value to get it to 0mw i believe. But would be weird if they didnt map all the values to power levels thats pretty [...]

Can you make a custom firmware build for me to give It a try? Maybe It will help other guys with the same vtx

closed 2021-06-10 by stale[bot]