FlightOne/FalcoX-Issue-Tracker #222: [NEW TARGET] New Target For Firmware

binarylefmart, 2021-05-03:
Hi guys,
Can you please build a Matek f405 target ? 
Here is specification:

MCU: 168MHz STM32F405RGT6
IMU: 32K ICM20602 gyro/accelerometer (SPI)
Baro: BMP280 (I2C)
OSD: BetaFlight OSD w/ AT7456E chip
Blackbox: MicroSD card slot (SD/SDHC)
VCP, UART1, UART2, UART3, UART4, UART5
Built in inverter for SBUS input (UART2-RX)
PPM/UART Shared: UART2-RX
SoftSerial on TX2, S5 or S6 optional
Camera control on S6 or DAC optional
Smartaudio & Tramp VTX protocol supported
Battery Voltage Sensor: 1:10
Current Sensor: No (FCHUB-6S, FCHUB-VTX, FCHUB-W option)
BEC 5V: No (FCHUB-6S, FCHUB-VTX, FCHUB-W option)
LDO 3.3V: Max.300mA
I2C1 SDA & SCL: Yes
WS2812 Led Strip : Yes
Beeper : Yes
RSSI: Yes

3x LEDs for FC STATUS (Blue, Red) and 3.3V indicator(Red)
6x PWM / DShot outputs without conflict
2x 2812LED outputs option
5x UARTs
1x Group of 5V/G/S1/S2/S3/S4 pads for 4in1 ESC Signal/GND
4x pairs of corner pads for ESC Signal/GND connections (DSHOT compatible)
1x pair I2C1 pads
1x Side-press button for BOOT(DFU) mode
1x 16pin bottom mounted FFC Slot for FCHUB-6S, FCHUB-VTX or FCHUB-W connection
36x36mm PCB with 30.5mm mounting holes
w/ 2x 0.5mm*16Pin 5cm Flexible Flat Cable
w/ 4pcs M3 Anti-vibration Standoffs

BTFL TARGET : MATEKF405

CoderElectronics, 2021-05-23:
I believe this is what you are looking for: https://flightone.com/firmware/alpha/FLX_MAMBAF405US_10.1.0.5490.flx

WildWillyFL1, 2021-07-26:
Looking at this target - is it the MATEKF405STD, MATEKF405SE, MATEKF405MINI, MATEKF405CTR, or MATEKF405AIO? Betaflight has these 5 different targets up as options for MatekF405