FlightOne/FalcoX-Issue-Tracker #226: target BETAFPVF405 problem with AIO Toothpick F405

DehtyarP, 2021-05-12:
Hello,

I need your kindly support, I have an issues with target BETAFPVF405 on my AIO  Toothpick F405, that AIO use target BETAFPVF405 with BlHelli_32. [](https://betafpv.com/collections/brushless-flight-controller/products/toothpick-f405-2-4s-aio-brushless-flight-controller-20a-blheli_32-v4)

But when i flash latest Version 10.1.0.5490 |10.1 Alpha 27| by Configurator 10.1.8-alpha. i have no OSD in any format(PAL,NTSC), and lose 2 ESC(works only 2).   When i return to latest BF OSD and all ESC works good.

Please check and fix it.

[image]
[image]
[image]
[image]
[image]

y0zean, 2021-05-31:
I just bought a license and I have the exact same issue. 
Doesn‘t find my tramp VTX and two motors won‘t spin during setup.
After flashing Emu or Betaflight it works again.

DehtyarP, 2021-07-28:
any news about it?

PGIII, 2021-08-04:
> any news about it?

@WildWillyFL1  Is working on a fix for this likely something set wrong in the target on our side