FlightOne/FalcoX-Issue-Tracker #227: FC not detected after changing target  & flashing via alpha configurator

dustyprops, 2021-05-14:
Configurator - Alpha 10.1.8
FC - CrazyF411
Wrong flashed target - MATEKF411
Correct(changed) Target - MATEKF411RX

Only one red LED blinking when usb connected.

PGIII, 2021-06-01:
It sounds like you registered it as the wrong target. You can Change that in the configurator if you go to flashing then bootloader flashing and pick the change target button

dustyprops, 2021-06-01:
Yes, I noticed this solution shortly after I sent this email. But after I
change the target, my board was not detected anymore. No blinking LED on
the FC, just solid LED.

dustyprops, 2021-06-02:
Yes, I noticed this solution shortly after I submit this issue. But after I change the target, my board was not detected anymore. No blinking LED on the FC, just solid red LED.

PGIII, 2021-06-04:
You can always flash in DFU mode what target did you change it to and what is it supposed to be ? also what is the serial number of the board

closed 2021-07-16 by stale[bot]