FlightOne/FalcoX-Issue-Tracker #230: Iflight f722 TwinG - doesnt detect SBUS

HrrrPippi, 2021-05-31:
**Version**
FLX_IFLIGHTF722_TWING_10.1.0.5490

**Target**
Iflight F722 TwinG mini (20x20 version uses same target as the 30x30 version)

**Describe the bug**
I got FalcoX installed on the FC, but it doesn't auto detect VTX, which I can solve through the configurator, but it also doesnt detect the SBUS receiver i have installed. 

I even resoldered the receiver to R1 in order to fix any issues in that regard. But it didn't help. Receiver works in Betaflight.

**To Reproduce**
Flash an iflight F722 TwinG with the last build of falcoX and go through the wizard

Any help is appreciated.

ocservo, 2021-06-01:
I have the same problem with my iflight f7

HrrrPippi, 2021-06-01:
I got through radio wizard by typing resetconfig in the cli on my computer. But once I got to the motor wizard, once again, it didn't work.

ocservo, 2021-06-02:
Thanks, I'll try

HrrrPippi, 2021-06-02:
Can some on at flight one maybe work on this. I payed money for a licence. I think they should fix it or give me my money back.

ojwright845, 2021-06-02:
Same problem detecting frsky receiver. Anyway to get a refund?

HrrrPippi, 2021-06-02:
Yeah, I tried sending an email, no respons. I tried on Facebook, no response. Heck, betaflight has better customer service, and it is free.

HrrrPippi, 2021-06-03:
I asked for a refund through PayPal, and it went through. 

I would have preferred they fixed the problem instead. But looks like no one at flight one team are following up on and fixing the problems posted on this github.

PGIII, 2021-06-03:
This issue tracker isnt for instant support its so we can better keep track of issues and things that need to be fixed, we check it pretty frequently but some days are busier than others.

What issue did you have with the motor wizard ? it seems like the reason you originally opened this issue for was fixed with a config reset

PGIII, 2021-06-09:
Closing Due to inactivity and being a duplicate of #179

closed 2021-06-09 by PGIII