FlightOne/FalcoX-Issue-Tracker #232: [NEW TARGET] New Target mambaF722_I2C

jimburroni, 2021-06-02:
Hello! Visiblement les nouvelles versions de mamba f722 ne fonctionnent pas avec la cible mambaF7.
Serait il possible d'ajouter cette nouvelle carte dans les cibles? Par avance, merci!

jimburroni, 2021-06-19:
Bonjour, comment savoir si notre demande est bien prise en compte ? Par avance, merci

WildWillyFL1, 2021-07-26:
The current MambaF722 target will work fine with boards that use the MambaF722_I2C for now

jimburroni, 2021-07-26:
it doesn't work for mine.  has there been an update?

WildWillyFL1, 2021-07-26:
> it doesn't work for mine. has there been an update?

what doesnt work exactly?

jimburroni, 2021-07-26:
the installation goes well but the wizard is going around in circles and the radio is not detected.  RX and radio paired well.  I restarted the procedure with the cli commands but nothing.

WildWillyFL1, 2021-07-26:
> the installation goes well but the wizard is going around in circles and the radio is not detected. RX and radio paired well. I restarted the procedure with the cli commands but nothing.

so it boots up fine, it just does not detect the rx?

jimburroni, 2021-07-26:
> > the installation goes well but the wizard is going around in circles and the radio is not detected. RX and radio paired well. I restarted the procedure with the cli commands but nothing.
> 
> so it boots up fine, it just does not detect the rx?
> 

Yes that's right.  obviously I am not the only one to encounter the same problem with the new versions of these FC.

WildWillyFL1, 2021-07-26:
What RX are you using? And which UART is it hooked up to? The I2C and normal version of this target are 100% identical except for some I2C defines (which we dont even use so it doesnt matter), so the standard target should work the same on any board that uses the I2C target as well.

jimburroni, 2021-07-26:
> What RX are you using? And which UART is it hooked up to? The I2C and normal version of this target are 100% identical except for some I2C defines (which we dont even use so it doesnt matter), so the standard target should work the same on any board that uses the I2C target as well.
> 
> EDIT: Ok, seems that they actually reversed the motor order, so that will have to be changed, but the RX should still detect as the UARTs are identical. I will get a target up with the new motor mapping as well

my rx is an R9MX in F.Port.  for Uart, I have to look on betaflight because I do not know since the time.

WildWillyFL1, 2021-07-26:
have you confirmed that the rx is working properly and connects to another fc or in betaflight?

jimburroni, 2021-07-26:
> have you confirmed that the rx is working properly and connects to another fc or in betaflight?

Yes, I fly the drone under betaflight without any problem.

WildWillyFL1, 2021-07-26:
Can you try flashing the board through the alpha configurator and seeing if the rx gets detected then? Also if you can, let me know which UART the rx is hooked up to