FlightOne/FalcoX-Issue-Tracker #233: Unexpected filesafe with CRSF when RFMD switched 2 -> 1

Exception0D, 2021-06-04:
**Version**
FalcoX v10.0.18.4939

**Target**
FC REVOLT OSD

**Describe the bug**
Unexpected filesafe nearly in 10 meters away from start
CRSF receiver is fine, I was flying this setup on FL1 Revolt v1.2.94 without any issues before, until I tried Falco X
The filesafe has occurred right at that moment  when RFMD switched from 2 -> 1 (150Hz -> 50Hz).

**To Reproduce**
1. install FalcoX v10.0.18.4939 on FC REVOLT OSD with CRSF receiver
2. on transmitter: set dynamic power off, max power = 25mW
3. takeoff the kwad and fly slowly close to the ground until RFMD switched from 2->1
4. filesafe nearly in 10..20 meters away from start (depends on antenna orientation)

**Expected behavior**
do not filesafe. 

**Additional context**
please note: the filesafe does not occurred on FL1 Revolt v1.2.94

PGIII, 2021-06-07:
Sounds like the CRSF is taking a while to get new packets out when it switches speed, but try on the latest alpha with RxFPS on the screen and see if you can get a DVR of it happening, also does the FC show Failsafe in the OSD when it happens ?

Exception0D, 2021-06-12:
Yes, FC shows "filesafe" when RFMD switched between 2 and 1.
This issue was fixed in alpha firmware (Version 10.1.0.5414 |10.1 Alpha 20|), I tried alpha and confirm: the issue with CRSF receiver is not reproduced anymore.
But I can't trust "alpha" due to unexpected disarm during flight:
https://www.facebook.com/groups/flightone/permalink/4379087845476317/

When all motors goes down I saw warning message: "throttle down" or something like that.

PGIII, 2021-06-13:
What Version of crossfire are you using

Exception0D, 2021-06-14:
I've tried crossfire v3.78, and latest v4.11, the filesafe reproduced on FalcoX v10.0.18.4939 for both.
For FalcoX v10.1.0.5503 |10.1 Alpha 28| the issue with receiver is not reproduced no matter what version of crossfire I'am using.

PGIII, 2021-06-15:
Do you have Blheli Telemetry Enabled ? try disabling that if you do

Exception0D, 2021-06-16:
yes, the BLheli telemetry is enabled. I wish to have an actual amp draw / current on OSD, but seems this feature is not working even for genuine flightone Bolt32 ESC. The BLheli telemetry works well on FlightOne v1.2.94
Thanks for advice, will try to disable BLHeli telemetry.

closed 2021-07-23 by stale[bot]