FlightOne/FalcoX-Issue-Tracker #241: h7 reboots in flight for no reason, with firmware higher than 10.1.0.5186

michael9544, 2021-07-12:
h7 reboots in flight for no reason, with firmware higher than 10.1.0.5186

H7 + TBS Tracer

PGIII, 2021-07-24:
Do you have a dvr of the issue and what fw version of tracer are you using

michael9544, 2021-07-24:
No dvr, the fc reboots and the vtx is off. 
Tracer 5.07

Mit freundlichen Grüßen

Michael Falke

Tischlerei Falke
Molkereistrasse. 5
29393 Groß Oesingen

Fon:    05838 - 268
Mobil: 0171 - 1405386
Fax:    05838 - 1208
E-mail: ***@***.***

Diese E-mail wurde vom iPhone gesendet.

PGIII, 2021-08-03:
There have been some recent changes to CRSF protocol in alpha 31 so try that and see if you still have the issue, try to get a dvr if possible

michael9544, 2021-08-04:
The newest version works fine.
Thank you

closed 2021-08-04 by PGIII