FlightOne/FalcoX-Issue-Tracker #242: [FEATURE] Active buzzer for T Motor F7 hd

ANTADRONEUR, 2021-07-16:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

PGIII, 2021-07-16:
Can you please provide more detail

ANTADRONEUR, 2021-07-16:
Hi
I have a t motor f7 hd target on Falcox licence Alpha, the 3 wires autonomous Vifly buzzer solder on b- (buzz) 5v and gnd on the fc 
On Falcox configurator when you choose a <buzz on > or <pintoogle > with switch affected , the buzzer does not work, like as if the buzzer was disabled on this flight card target
A lot of people are the same issue with this target on Falcox Alpha licence buzzer don’t work with a radio switch

Regards

Eric DESPLACES

PGIII, 2021-07-24:
Ah ok we’ll have have to take a look at the target could be something set wrong

ANTADRONEUR, 2021-07-24:
Hi

Ok thank you, let me know if you solve the buzzer issue with this target.

Bubu38, 2023-12-08:
hi i have the same issue with t motor f722 AIO 35A under BF 4.4.3 buzzer not working a t all ...