FlightOne/FalcoX-Issue-Tracker #244: [FEATURE] New Feature For Firmware.  Enable blackbox logging

RoninMedia12, 2021-07-28:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

i really wish i could use blackbox to tine with different builds have strange issues being able to read some blackbox data to pin point what freq i need to target. 

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

PGIII, 2021-08-04:
Data Logging support is planned but can't give a timeline on it right now

RoninMedia12, 2021-08-05:
Thank you for your response, and update on roadmap.