FlightOne/FalcoX-Issue-Tracker #245: Endpoints from Radioconfig are flawed since  10.1.1.5664

michael9544, 2021-08-03:
H7

PGIII, 2021-08-03:
I don’t really understand what this means what issue are you having?

michael9544, 2021-08-03:
The endpoints Min + mid + maxrc,
the values ​​are not interpreted correctly.

PGIII, 2021-08-03:
I don’t understand why do you think it’s not correct ? What issue are you having?

michael9544, 2021-08-04:
I Look into Systeminfo.

Mid position on al channels by 2%
min and max have 100% by half stick movement.

PGIII, 2021-08-04:
What Rx and radio are you using ?

michael9544, 2021-08-04:
Tracer + Mambo

PGIII, 2021-08-04:
What version is on Tracer also have you tried running RESET_WIZARD RADIO in the console and going through the radio setup again

michael9544, 2021-08-04:
6.07

PGIII, 2021-08-04:
Going through the Setup again fixes it for me, looks like its due to CRSFv3 having a higher resolution than before

michael9544, 2021-08-04:
i try RESET_WIZARD RADIO.

in systeminfo:
sometimes value goes correct from-100 to +100
after reboot
sometimes goesfrom from -82 to +76

minrc=0
midrc=1022
maxrc=2047

michael9544, 2021-08-04:
i have no problems with 10.1.0.5186

minrc=175
midrc=992
maxrc=1809

PGIII, 2021-08-04:
Its due to CRSFs changing resolution ill have to look into normalizing all of them into the same range

PGIII, 2021-08-04:
> i have no problems with 10.1.0.5186
> 
> minrc=175
> midrc=992
> maxrc=1809

that version doesn't support the dynamic resolution with CRSF v3 which is why

PGIII, 2021-08-04:
[CRSF_Scaling_FixLIGHTNING_10.1.1.5667.flx.zip](https://github.com/FlightOne/FalcoX-Issue-Tracker/files/6933374/CRSF_Scaling_FixLIGHTNING_10.1.1.5667.flx.zip)
try this, you'll probably need to redo the radio wizard

michael9544, 2021-08-04:
I have now flown 4packs.
no problems so far.
feels even better than 10.1.0.5186.
good job, thank you very much

closed 2021-08-04 by PGIII