FlightOne/FalcoX-Issue-Tracker #247: H7 Quopa and Motorbeep on same switch, quad freaking out

michael9544, 2021-08-04:
H7 
redux32

dshot 64

I have Quopa and Motorbeep on same switch,
When new arm, sometimes the motor direction is wrong and quad freaking out

PGIII, 2021-08-04:
They Both send Commands to the motors, could maybe add a latching to the motor beep mode so it needs to be on for a period of time before sending the commands, what position on the switches is each mode on ?

michael9544, 2021-08-04:
quopa and motorbeep are on the same switch position

PGIII, 2021-08-04:
Ah well they can't be on the same position they both send commands to the ESCs so they wouldn't be able to do that at the same time

michael9544, 2021-08-04:
is it possible on on switch ?

down - nothing
mid -launch + motor beep 
up- quopa

closed 2021-08-05 by michael9544