FlightOne/FalcoX-Issue-Tracker #249: [NEW TARGET] New Target For Airbot Omnibus F4 Nano Size 20x20

saptana26, 2021-08-06:
hello everyone, does anyone know that the fc Airbot Omnibus F4 Nano Size 20x20 is already supported by flightone. Yesterday, I tried flashing the firmware with omnibus f4sd and omnibus f4 according to the store's recommendation, selling the fc and buying a license. but after successfully installed, I always stop at the wizard when detecting the receiver. other than that when I try to turn on google, the osd writing also doesn't appear. does anyone know my problem?

[image]

WildWillyFL1, 2021-08-07:
What target does this FC use if you flash Betaflight?

saptana26, 2021-08-07:
> What target does this FC use if you flash Betaflight?

I use OMNIBUSF4NANOV7 on betaflight