FlightOne/FalcoX-Issue-Tracker #251: Version 10.1.1.5677 |10.1 Alpha 33|| Matek F722-SE

DoomedFPV, 2021-08-08:
**Version**
Version 10.1.1.5677

**Target**
Matek F722-SE

**The bug**
Unable to detect radio.

Running a CRSF setup but unable to get pass the Detect Radio Wizard part of the setup.
I've tried setting the Serial ports manually, changing the max baud speed upto 115200 and swapping Pins but still nothing.
I've confirmed my CRSF is set to the correct UART.

**To Reproduce**
Just flash the FC and run the wizard

**Expected behavior**
To detect radio

**Additional context**
Had the same bug or a previous version.  ( Version 10.1.0.5490 |10.1 Alpha 27 )

PGIII, 2021-08-08:
There’s an issue open already for this I believe #190, But try RESETCONFIG  in the console, could be something set wrong in the target that we may need to check.

DoomedFPV, 2021-08-09:
#190 I followed it and was hoping the issue was fixed in the latest version as I've not seen any further comments or issues.

I've reflashed the FC and it made no difference.
I've issued the RESTCONFIG command I can see the command was accepted and the config was reset, but still no difference.

Is there any log file, I can upload to help?

WildWillyFL1, 2021-08-12:
What UART is CRSF on? And can you try setting the CRSF RX to SBUS or n.i. SBUS and see if it detects?

DoomedFPV, 2021-08-12:
No Joy.
I tried SBUS and n.i. SBUS with and without wire switching, with doing RESETCONFIG between each change.
I also updated to 10.1.1.5689 and tried again with no luck.

My UART Port layout is:
UART 1 - Empty
UART 2 - CRSF
UART 3- Empty
UART 4 -Empty
UART 6 - VTX (TBS Smart Audio)

PGIII, 2021-08-13:
[WIZ_INFO_MATEKF722SE_10.1.1.5697.flx.zip](https://github.com/FlightOne/FalcoX-Issue-Tracker/files/6984153/WIZ_INFO_MATEKF722SE_10.1.1.5697.flx.zip)
Try this version and get a video or DVR of the wizards

DoomedFPV, 2021-08-13:
During the Wizard my transmitter was switched powered on and bound to the RX and was setup for crossfire.
I also had a lipo plugged in and made sure  the VTX was powered up and outputting to my goggles..

But still no joy YT link below.  

https://youtu.be/Tzms8-v2fN4

PGIII, 2021-08-13:
[WIZ_INFO_MATEKF722SE_10.1.1.5703.flx.zip](https://github.com/FlightOne/FalcoX-Issue-Tracker/files/6984610/WIZ_INFO_MATEKF722SE_10.1.1.5703.flx.zip)
Try getting a Video with this

DoomedFPV, 2021-08-13:
OK some success :)
It didn't detect, then I did a RESETCONFIG and it picked up skip to 5:52 on the YT....
I ran through the rest of the wizard but my bench lipo was too dead to spin the motors, But the wizard did finish.
I tried again with a more charged lipo, but it didn't detect...
Charging a lipo and trying again.

https://youtu.be/BiUhyDRpp7A

DoomedFPV, 2021-08-13:
Success....
It took a couple of RESETCONFIGs but after a reboot caused by the wizard after the search for External OSD it detected and completed the wizard.

The video shows it searching, but at 4:40 after the reboot, it detects the RX and continues.
https://youtu.be/Epc2qm16FPc

Thanks @PGIII 
If you want me to try any other versions, just let me know. Happy to help.

PGIII, 2021-08-13:
Hmm that's really Strange, do you have the RX and TX connected to the board on the Crossfire ?

DoomedFPV, 2021-08-13:
Yeah
Works fine every time in BF.

Now that I've repowered the quad, it looks like I've lost the RX again.
Tried power cycling a couple of time, but nothing.

[image]

DoomedFPV, 2021-08-14:
Something I've just noticed.
That when I plug in a lipo, I only get the first 3 FC tones.
When it worked and detected the RX I got the full set of FC tones.

DoomedFPV, 2021-08-14:
I've tried RESETCONFIG and now its not detecting the RX again.
I've tried unplugging my VTX on UART6 in case its having an effect.

But I did notice, the Wizard is detecting my RX.....
At the start of the video my TX is turned off and the errors increase slowly across all the UARTs, But then when I turn it on UART 2 errors climb.  It almost looks like its detecting my RX but doesn't like the data coming from it?

https://youtu.be/0ZFAZYNc8nc

PGIII, 2021-08-14:
[GYRO_FIX_MATEKF722SE_10.1.1.5703.flx.zip](https://github.com/FlightOne/FalcoX-Issue-Tracker/files/6986769/GYRO_FIX_MATEKF722SE_10.1.1.5703.flx.zip)
Try this

DoomedFPV, 2021-08-14:
@PGIII 
I think that got it this time,
Lipo plugged in, TX switched on and it detects and completes the wizard.
I've done a RESETCONFIG twice, and lots of power cycles and it works every time.

Well done sir and thank you

PGIII, 2021-08-14:
Glad to hear will update that in the next alpha

closed 2021-09-13 by PGIII