FlightOne/FalcoX-Issue-Tracker #252: [NEW TARGET] New Target For Firmware

CEaston72, 2021-08-15:
Hi
  Please could you add a target for the Matek F722 mini SE

WildWillyFL1, 2021-09-16:
Can you check and see if the board is running the MATEKF722SE or the MATEKF722MINI target?

CEaston72, 2021-09-16:
Hi Willy
 Thank you for picking up this request. I‘m currently running the following:
# version
# Betaflight / STM32F7X2 (S7X2) 4.2.9 Apr 27 2021 / 19:34:29 (e097f4ab7) MSP API: 1.43
# config: manufacturer_id: MTKS, board_name: MATEKF722MINI, version: 55694498, date: 2021-05-15T13:56:51Z
# board: manufacturer_id: MTKS, board_name: MATEKF722MINI
With everything on the quad working 
The manual (http://www.mateksys.com/?portfolio=f722-minise#tab-id-3) does state that you can also use the MATEKF722SE legacy target. However since I have got my board Matek have brought out the F722SE as an AIO board so I’m not sure if this information is still relevant ?
The last version of FalcoX I tried was the MATELF722SE Version 10.1.1.5717 |10.1 Alpha 36| all flashed ok, and apart from no VTX being found I was able to complete the configuration wizard on the PC and Transmitter. 
On trying to workout what was going on with the VTX I went through the settings on the configurator making sure the serial ports was setup and in the vtx tab was set to TBS nano32 still with no luck. I then used a meter and found that there was no power being sent to the VTX or the camera? At this point I thought it was a firmware mismatch possibly even a pinio problem but I don’t know enough about this to say :-/ 
I have now gone back to the above version of Betaflight and everything is back working fine, and at no point have I had to change any of the wiring around.
After finding that there was no power, I thought it was time to reach out to you guys

WildWillyFL1, 2021-09-17:
Yeah it sounds like its a VTX power switch issue, I can get you a test bin if you are back on FalcoX. The Mini and SE targets are identical though, I'm just guessing on the Mini they use the PINIO specifically for an internal vtx pit switch.

CEaston72, 2021-09-17:
That would be great, I can reflash falcox back on this model and if you can tell me what to do from there, it would be good. As I still want to see how this model flies on FalcoX :-)