FlightOne/FalcoX-Issue-Tracker #256: H7 Quapa

michael9544, 2021-08-21:
all version

if the Motobeep was used after quopa, the engine sometimes turns in the wrong direction

PGIII, 2021-08-21:
how soon are you using motor beep after ?

michael9544, 2021-08-22:
shortly after.
switch down, off
switch middle, beep
switch up, quopa

PGIII, 2021-09-13:
> shortly after.
> switch down, off
> switch middle, beep
> switch up, quopa

They both send commands to the motors so you probably need to wait a few seconds between using the two modes, we could probably add something to protect you from doing it to quick although the timing is likely different between ESCs. But you can also change the order so that the Beep is the end position for now so that you aren't going past it to use quopa mode, or use a different switch

michael9544, 2021-09-13:
Thank you.

unfortunately channel 8 is permanently occupied by rssi.
if channel 8 were freely usable, it would make things easier.

closed 2021-10-21 by stale[bot]