FlightOne/FalcoX-Issue-Tracker #257: BLADEF7 Buzzer not working

V1NCEFPV, 2021-08-22:
**Version**
10.1.0.5252

**Target**
BLADE_F7

**Describe the bug**
Buzzer not running, motor beep is ok.
I tried to set the buzzer via configurator and osd.
Still no sound.

**Expected behavior**
Buzzer working

MakeDa002, 2021-09-05:
What type of buzzer are you using?

V1NCEFPV, 2021-09-06:
I'm using a Vifly finder. (3 wires )

MakeDa002, 2021-09-06:
Do yo have the buzz- connect to tx4?

V1NCEFPV, 2021-09-06:
No, TX4 is assigned to VTX by default on the fast connection.

V1NCEFPV, 2021-09-06:
The buzzer is connected to Buz+ Buz- pads and a GND

MakeDa002, 2021-09-06:
There is a post with 3 wires buzzer connect. https://support.flightone.com/index.php/knowledge-base/connecting-a-buzzer/

I use the revolt osd, and the buzz- of the buzzer isn't connecting to buzz- pad. And mode page of OSD you have to configure pintoggle, not use BuzzOn. 

Try connecting buzz- to buzz- or connect buzz- to other tx. 

Good look!

V1NCEFPV, 2021-09-06:
OK, thanks. I'll give a try and keep you up to date.

shadowbit23, 2021-11-01:
i still have the same issue buzzer doesn' working even if i connect to another port and use pintoggle mod