FlightOne/FalcoX-Issue-Tracker #261: Alpha 37 osd problem

diankurus, 2021-08-29:
**Version**
10.1.1.5721

**Target**
Revolt Osd & FuryF4osd

**Describe the bug**
Osd bug. Flickering on menu and home screen / osd screen both revoltosd and furyf4osd

diankurus, 2021-08-31:
I tried to flash to stable firmware to make sure its not hardware problem and yeah osd is good on stable version

PGIII, 2021-08-31:
Do you have a video of the issue?

diankurus, 2021-08-31:
I cant post video. Since flickering is match with video fps so in video wont be any osd flickering. In dvr its show no osd . But there're osd!

[Uploading 131506760...]

diankurus, 2021-08-31:
Updates :

Osd works fine on caddx ratel

Osd flickering on runcam racer 3.

5v was feed from vtx before. Then 5v from fc .

diankurus, 2021-09-01:
so it was solved.

My Runcam use NTSC as default. falcox video system in auto didnt work. 
my bad. i did try to change on falco video system to NTSC and hit save and exit without power cycle so make no difference

now tried to use pal on my runcam my osd is good
change runcam to ntsc and change falco video system to NTSC and hit save and do powercycle OSD is good

PGIII, 2021-09-01:
Ah the OSD chip must just have issues detecting which mode that camera is on, glad you figured it out

closed 2021-09-01 by PGIII