FlightOne/FalcoX-Issue-Tracker #264: [NEW TARGET] New Target For Aikon F7 v2 and Aikon F7 mini V3

TcheTcheFPV, 2021-09-07:
Ii would like to flash FL1 on :

Aikon F7 V2
and
Aikon F7 mini v3

Thanks :)

WildWillyFL1, 2021-09-16:
I believe those both use the AikonF7 target which is already supported in the current alpha releases. To make sure you can plug those FCs in betaflight and confirm they use the AIKONF7 target

TcheTcheFPV, 2021-09-16:
Hi @WildWillyFL1, thank you for the answer.

So, I've checked on BF releases and it seems that they provide AIKONF7 firmware for all Aikon F7 boards, so you might be right.
Unfortunately I will not be able to check that because I decided to go for a Hobbywing F7 which is supported.

Thanks to your input, I understand that different FC's can have the same firmware target.