FlightOne/FalcoX-Issue-Tracker #265: [NEW TARGET] New Target For Kakute F7 mini V2

TcheTcheFPV, 2021-09-07:
After having successfully flashed and tuned the last alpha released on a kakuteF4V2, I'm looking forward to also flash it on the KAKUTE F7 MINI V2

Thanks :)

TcheTcheFPV, 2021-12-09:
> Will do, shoot me a message on Facebook if you have me added, thats a bit easier for quick debugging than the GitHub page.

Haven't seen your answer till now, sorry. OK I will connect you on FB

TcheTcheFPV, 2022-05-01:
Hi @WildWillyFL1 ,
Do we have any news regarding F745 support on Kakute F7 board ?
For the moment, the board isn't booting after updating the bootloader.

TcheTcheFPV, 2022-06-07:
Hi @WildWillyFL1 , did you get my last message ?
Is there any chance to get F745 boards supported ?
Actually the bootloader is not booting, but I can do some test for you on my side if you want.
Please let me know

TcheTcheFPV, 2022-06-27:
@WildWillyFL1 @PGIII please let me know.

TcheTcheFPV, 2022-07-07:
Hi @PGIII @WildWillyFL1 

I've seen that you have just published an update for F745 support. So I tried on my kakute F7miniV2 and it still has issue : this time, after updating the bootloader, the FC is always booting in bootloader mode.
After updating the bootloader, it shows an error : "An error has occured : Updating : Drive path undefined."

[image]

At this stage, I have tried to flash the firmware (and it asked for a license code, that I have filled in), but I think it flashes the firmware into the bootloader memory, because the FC is then not booting anymore (Need to boot while pushing the boot button on the FC to get it back again booting in bootloader mode).

It's nearly here :)

PGIII, 2022-07-07:
I'm working on fixing the F745, there'll be some more updates in the next couple days, bootloader working is a good first step :)

TcheTcheFPV, 2022-07-08:
Hi @PGIII,

I have seen the last update (10.2.3 |10.2 Alpha 4|). Here is the feedback :

first, It could flash the bootloader and reboot correctly, showing "kakuteF7v2 in bootloader mode". Then flashing the firmware was working, but then, once the FC was power cycled, it shows again "kakuteF7v2 in bootloader mode". 

After some try, it now shows "F745_blflash In BootLoader Mode" once the bootloader updated, and don't let me update the firmware automatically (saying that the target is not found, because of the name I guess), and even not manually also (staying at "flashing 0%".

PGIII, 2022-07-08:
Put the board into DFU mode and get a video of what it does when you flash in DFU mode

TcheTcheFPV, 2022-07-08:
Sure, here it is : https://youtu.be/kQnHu5MAF_s

PGIII, 2022-07-08:
You don't need to pick change target in the first part thats just for changing what the board is registered as in the system. The BLFlash part is normal The F7 and H7 don't have unique serial numbers in DFU mode so that is used to get the serial number. I forgot I had made some changes to the Configurator ill have to upload a new version that should make it auto flash at that point like it should

PGIII, 2022-07-08:
I just pushed a new configurator try flashing in DFU mode with that one

TcheTcheFPV, 2022-07-09:
I have downloaded and installed the last configurator, and even if the bootloader is automatically updated, it still shows "F745_blflash In BootLoader Mode" and then I can't update the firmware (like in the video), it says "Target not found", and if I try manually, it stays at 0% (even waiting half an hour to be sure).

What is strange is that I remember yesterday, at my first try, I could update the bootloader, and it showed "kakuteF7v2 in bootloader mode". then I could update the firmware, and I saw it launching the UARTs detection in the OSD. But when I power cycled the FC, it was coming back in bootloader mode.
So I tried again 2 or 3 times, and suddenly, it start to show "F745_blflash In BootLoader Mode" after bootloader updating.
Since, I can't flash the firmware.

PGIII, 2022-07-09:
It should DFU flash twice, it needs to dfu flash again when it has the blflash bin loaded that's just for getting the serial number, could you get a video of flashing in DFU mode with the new configurator so I can see what's happening

TcheTcheFPV, 2022-07-09:
Ah ok, in this case, the FC is not booting after flashing the Firmware at the second DFU connection. Video is coming.

TcheTcheFPV, 2022-07-09:
here is the video : https://youtu.be/rNzi8M2Cp50

PGIII, 2022-07-09:
The first time it shows the DFU page you can just pick update to start flashing, it should auto flash when it shows BLflash fw I'm not sure why it's not for you. But now that it's flashed try unplugging and plugging it back in 5 times in a row quickly and see if it comes up in bootloader mode

TcheTcheFPV, 2022-07-09:
The "5 times" trick worked (didn't know about it). So, I could flash it perfectly.

Now, it is at the "detect radio wizard" stage.
It detects my VTX immediately, but not my RX (crsf). The RX is on uart6 (USART6). Is it a problem ?
(Radio is on and RX is binded).

TcheTcheFPV, 2022-07-09:
And also, the FC is booting with 3 beep and not the 5 as usual, and shows the wizard;

TcheTcheFPV, 2022-07-09:
I've moved the CRSF on UART4, and still not detected (it's working with betaflight so the RX is not broken).

There is something else, I don't know if it means something but, the "5 times" trick is a 3 time trick in reality. 3 times unplug/replug the FC is enough to get it back working after bootloader flashing.

PGIII, 2022-07-09:
Ok that might be something set wrong in the target, and it is something like 3 times I usually tell people more to be sure they do it enough times :)

TcheTcheFPV, 2022-07-11:
Hi @PGIII ,

Feedback from the last version (10.2.4 |10.2 Alpha 5|) : The ESC is still starting with only 3 beep instead of 5, and the radio wizard does not detect the RX. The VTX is still well detected though.

PGIII, 2022-07-14:
> Hi @PGIII ,
> 
> Feedback from the last version (10.2.4 |10.2 Alpha 5|) : The ESC is still starting with only 3 beep instead of 5, and the radio wizard does not detect the RX. The VTX is still well detected though.

Ok ill have to check the defines for that target we must have something set wrong, what Gyro does the board have ?

PGIII, 2022-07-17:
Hmm ok I'll have to double check the target is set to use those gyros

PGIII, 2022-07-18:
[FLX_KAKUTEF7V2_10.2.5.6070.flx.zip](https://github.com/FlightOne/FalcoX-Issue-Tracker/files/9133385/FLX_KAKUTEF7V2_10.2.5.6070.flx.zip)
Try this version and see if it fixes the motors

TcheTcheFPV, 2022-07-18:
Hi, I've tried it and unfortunately it makes no difference.

TcheTcheFPV, 2022-07-23:
Hi @PGIII ,

I have given a try to the last alpha (Version 10.2.6 |10.2 Alpha 7|), but the issue is still here : only 3 beep when plugin the battery, and the crossfire receiver is not detected.

TcheTcheFPV, 2022-08-18:
Hi @PGIII ,

I can confirm that the gyro is a PMU6000 on this board 

# version
# EmuFlight / KAKUTEF7MINIV2 (KTF7) 0.4.1 Jul 29 2022 / 12:56:58 (7bac6ddd0) MSP API: 1.51

# status
System Uptime: 14 seconds
Current Time: 2022-08-18T09:00:30.841+00:00
Voltage: 1 * 0.1V (0S battery - NOT PRESENT)
CPU Clock=216MHz, Vref=3.30V, Core temp=40degC, GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
Stack size: 2048, Stack address: 0x20010000
I2C Errors: 4, config size: 2823, max available config: 32768
CPU:4%, cycle time: 125, GYRO rate: 8000, RX rate: 33, System rate: 9
Arming disable flags: RXLOSS CLI MSP

TcheTcheFPV, 2023-01-23:
Hi @PGIII , hope you're well

Do you have any news on this one ?

Thank you :)

PGIII, 2023-01-25:
Ill try to take another look at that target this week

PGIII, 2023-01-25:
I don't see anything obviously wrong in the target. Can you try flashing the latest Alpha and typing GYRO_DETECTED in the console and let me know what it replies with

TcheTcheFPV, 2023-01-28:
Hi @PGIII ,

I have flashed again the FC, and I still get only 3 beep at startup and the command replies #ERROR No Gyro Detected

[image]

[image]

TcheTcheFPV, 2023-03-15:
Hi @PGIII ,

Have you seen something in the target ?

TcheTcheFPV, 2023-03-27:
Hi @PGIII ,

Any news on that ?

Thank you :)

TcheTcheFPV, 2023-04-27:
Hi @PGIII ,

Have you find something ?

TcheTcheFPV, 2023-06-12:
Hi @PGIII , is there any news for that one ? Please ?