FlightOne/FalcoX-Issue-Tracker #268: [NEW TARGET] New Target For Holybro Kakute F7 v1.5

striker890, 2021-09-23:
I would like to use the Kakute F7 v1.5 (the one with the MPU-6000)
Would be awesome if you are able to create a target for this.

Jurisjury, 2021-09-24:
This was requested since February this year and no update and support.

WildWillyFL1, 2021-09-26:
I started a target for this a while back but it uses some weird pin mappings we would need to add low level driver support for so it's more of a larger job than adding other targets. We may add it in alpha for testing if we get enough requests for it due to the amount of work needed to add it

saxman31, 2021-09-28:
Hi, i'm also interested to use FalcoX with my kakute F7 v1.5

TcheTcheFPV, 2021-10-06:
> I started a target for this a while back but it uses some weird pin mappings we would need to add low level driver support for so it's more of a larger job than adding other targets. We may add it in alpha for testing if we get enough requests for it due to the amount of work needed to add it

Hi @WildWillyFL1 , do you have the same thing with the kakute F7 MINI (V2) ?

TcheTcheFPV, 2021-10-11:
> Hey guys, I just realized that the Kakute F7 boards all use a different F7 processor than the other F7 targets we support so for now I don't think it will be possible but I'm talking with PG3 to see what the process would be of adding support for that MCU, but I would not count on it for now.