FlightOne/FalcoX-Issue-Tracker #270: GPS and ESC Telemetry Not Detecting

Icepik125, 2021-09-29:
**Version**
FLX_LIGHTNING_10.1.1.5739, although earlier versions did not work.  

**Target**
Lightning H7

**Describe the bug**
HGLRC M80 GPS Unit is properly connected and receives power.  When running Detect GPS, no GPS is detected.  Will time out back to Peripherals page (7/14), or OSD will say "NO GPS DETECTED".  Same behavior noted on UART 3 and UART 8.  Changing max baud rate does not fix.  Changing UART from "OFF" to "GPS" does not fix.  Confirmed that TX is connected to RX, and vice versa.  Changing to Matek M8Q-5883 generated same result, so I believe both GPS units are good. 

Similar behavior when attempting to detect ESC Telemetry.  ESC is powered up.  Select Detect BLHeli32 telemetry; will return an error "no telemetry detected".  

**To Reproduce**
Steps to reproduce the behavior:
1. Attach GPS, power on flight controller from 4S lipo
2. Open OSD
2. Navigate to Page 7
3. Scroll down to Detect GPS and select
4. See error

**Expected behavior**
GPS is detected.  FalcoX permits setup of return to home.

Icepik125, 2021-09-29:
To add:  I attempted the fixes from Issue 74 (similar GPS detection error) but still will not detect.

WildWillyFL1, 2021-10-10:
So to confirm, both HGLRC and Matek GPS units when hooked up were not detected?

closed 2021-11-17 by stale[bot]