FlightOne/FalcoX-Issue-Tracker #272: No current reading on XrotorF7

TcheTcheFPV, 2021-09-30:
**Version** 10.1.1.5728

**Target** XrotorF7conv

**Describe the bug**
There is no value on the "Amp draw" in the OSD. It stays at 0.
I'm using an ESC (blheli32) from Aikon (that model : https://www.getfpv.com/aikon-ak32-55a-3-6s-blheli32-4-in-1-esc.html), and both TELEMETRY and CNT from the ESC are connected to TELEMETRY and CRT on the flight controller. Voltage reading is ok.

I don't find a way to setup current source in the configurator. Is there any terminal command to setup CRT ?

**To Reproduce**
Steps to reproduce the behavior:
Connect a ESC with TELEM and CRT on a XrotorF7 board and try to get Amp draw in the OSD.

**Expected behavior**
It should show the Amp draw.

WildWillyFL1, 2021-10-01:
Do you have blheli telemetry set on any of the UARTs? If telem is set on any of the UARTs it will use the telemetry for current draw otherwise it will use the analog current. Let me know if there is no current being displayed with telem off and I can look at the current sensor on it.

TcheTcheFPV, 2021-10-02:
Hi @WildWillyFL1 and thank you for your answer.

OK, so, as this ESC does not send current information via TELEM, but only via analog CRT, I have to unset UART6 from BLHELI TELEM to get analog value reading, right ?

So, I have turned off blheli telem, and I can now get the current reading correctly.

So it's working now. Thank you.

TcheTcheFPV, 2021-10-02:
Thanks for your help :)

closed 2021-10-02 by TcheTcheFPV