FlightOne/FalcoX-Issue-Tracker #274: Incorrect LQ value on OSD when using ELRS

achmiral, 2021-10-05:
**Version**
10.1.1.5748

**Target**
Zeusf722aio and Revolt

**Describe the bug**
ELRS use AUX 11(CH15) as LQ channel and AUX12(CH16) as the RSSI channel [^1]. As I know FalcoX reads LQ from AUX4(CH8) so the value on the RSSI OSD element are incorrect and not represent the LQ value from ELRS which is on AUX11(CH15). The osd element value always change from 100% to 47%  back and forth.

**To Reproduce**
Steps to reproduce the behavior:
1. Install ELRS RX on the quad
2. Flash the FC using Alpha 10.1.1.5748
3. Power on the radio and the quad
4. Check the RSSI osd element on the screen

**Expected behavior**
The correct value of LQ from ELRS

**Screenshots**
https://www.youtube.com/watch?v=E_em3OAoir4
The RSSI OSD element is on the bottom right of the screen. 

**Additional context**
Add any other context about the problem here.

[^1]: https://www.expresslrs.org/1.0/info/signal-health/#rssi-value-osd-field

achmiral, 2021-10-15:
> Can the channels they output on not be changed ? It’s planned to allow the channel FalcoX search’s for RSSI on to be changed but that hasn’t been added yet

I think there is no option to change the output channel on ELRS right now, either using the configurator or using the lua script. Except changing directly to the source code and build the custom ELRS bin.

gretel, 2021-10-15:
this should be configurable in the flight control in any case. please do not ignore expresslrs just because it's open source.

gretel, 2022-03-29:
> > this should be configurable in the flight control in any case. please do not ignore expresslrs just because it's open source.
> 
> Like I said its a planned change since most newer protocols like SRXL2, GHST, and FPORT Send RSSI as part of the protocol and since we can get LQ through Telem on CRSF it would be good to be able to free up channel 8. Never said we are ignoring elrs because its open source

any news on this? it's getting a reason to leave falcox behind.

PGIII, 2022-03-29:
> any news on this? it's getting a reason to leave falcox behind.

Planned for next Alpha Release, What target are you using I can give you FW with support for it

gretel, 2022-03-29:
@PGIII `Mamba f411 AIO` - regards

PGIII, 2022-03-29:
[FLX_MAMBAF411_10.1.1.5906_RSSI_TEST.flx.zip](https://github.com/FlightOne/FalcoX-Issue-Tracker/files/8373982/FLX_MAMBAF411_10.1.1.5906_RSSI_TEST.flx.zip)
Here you go let me know if you have any issues, should be able to change in the configurator under Configuration>Serial or in the Text menu under the Serial Menu

gretel, 2022-03-29:
> [FLX_MAMBAF411_10.1.1.5906_RSSI_TEST.flx.zip](https://github.com/FlightOne/FalcoX-Issue-Tracker/files/8373982/FLX_MAMBAF411_10.1.1.5906_RSSI_TEST.flx.zip) Here you go let me know if you have any [...]

thanks, but can't set to `0` to disable it?

PGIII, 2022-03-30:
[FLX_MAMBAF411_10.1.1.5909_RSSI_TEST_2.flx.zip](https://github.com/FlightOne/FalcoX-Issue-Tracker/files/8381533/FLX_MAMBAF411_10.1.1.5909_RSSI_TEST_2.flx.zip)
Added an Option to disable RSSI and wizards should be able to see channel 8 now also

gretel, 2022-03-30:
> [FLX_MAMBAF411_10.1.1.5909_RSSI_TEST_2.flx.zip](https://github.com/FlightOne/FalcoX-Issue-Tracker/files/8381533/FLX_MAMBAF411_10.1.1.5909_RSSI_TEST_2.flx.zip) Added an Option to disable RSSI and wizards should be able to see channel 8 now also

[image]

@PGIII it's werking 😸 

noticed some expresslrs related cosmetical issues:

[image]

the `txpwr` value has `mW` as unit, not `dB`. could you please either make this configurable or remove the unit from the osd element?

[image]

`rxfps` is flickering from 500 to 1000 `fps` when it's locked at 500. while debatable the unit here is `Hz`.

PGIII, 2022-03-30:
> noticed some expresslrs related cosmetical issues:
> 
> [image]
> 
> the `txpwr` value has `mW` as unit, not `dB`. could you please either make this configurable or remove the unit from the osd element?
> 
I assume this is Using CRSF, the Telem data we get for that is converted to dBm so the value should be accurate, 10 mW happens to be the same as 10dBm. We could make the units selectable in the future, but I believe it was done that way to match what crossfire shows in the radio although I can't remember for sure

> [image]
> 
> `rxfps` is flickering from 500 to 1000 `fps` when it's locked at 500. while debatable the unit here is `Hz`.

RxFPS is measuring the time between control frames, hence "Frames Per Second", Hz would just be a more generic Unit. If it shows higher it means for some reason the FC is getting Control Frames quicker. Its common for CRSF to adjust the Frame rate based on Lq and other things but not sure what would make ELRS send it faster than what you're saying is set to. Does the TxFPS change?

gretel, 2022-03-30:
> I assume this is Using CRSF, the Telem data we get for that is converted to dBm so the value should be accurate, 10 mW happens to be the same as 10dBm. We could make the units selectable in the [...]

yes; ah i get it, so far no more than 10mW was required, so i also didn't see any value different than `10 dB`. can it be changed to `dBm`, though?
 
> RxFPS is measuring the time between control frames, hence "Frames Per Second", Hz would just be a more generic Unit. If it shows higher it means for some reason the FC is getting Control Frames [...]

elrs has no dynamic rate switching like tbs, but dynamic power. hence, the rate is locked at 500. betaflight, emuflight and inav show the value locked at 500 which i consider to be correct.

PGIII, 2022-03-30:
It can changed to dBm probably was a typo

PGIII, 2022-03-30:
>elrs has no dynamic rate switching like tbs, but dynamic power. hence, the rate is locked at 500. betaflight, emuflight and inav show the value locked at 500 which i consider to be correct.

https://prnt.sc/qc4UVP3VhQkv Took some Readings of the ELRS output on the receiver I have here and there is some variation in packets, every so often a packet comes in at 250hz instead of 500, so some flicker to a lower value in the FPS reading is possible, as far as it reading faster I'm guessing that is due to it being measured in Milliseconds, so it may read the time difference as 1ms instead of 2ms if the timing with the clock happens to work out that way. Externally measuring it looks like the FC is reading and processing the packets on time so I wouldn't worry about it

gretel, 2022-03-30:
> https://prnt.sc/qc4UVP3VhQkv Took some Readings of the ELRS output on the receiver I have here and there is some variation in packets, every so often a packet comes in at 250hz instead of 500, so [...]

but it's jumping from 500 to 1000.. there is no 1000Hz mode in elrs (yet). i'd be glad if the osd element could display values consistent with other flight control software parsing the same telemetry.

PGIII, 2022-03-31:
> but it's jumping from 500 to 1000.. there is no 1000Hz mode in elrs (yet). i'd be glad if the osd element could display values consistent with other flight control software parsing the same telemetry.

The time between the control packets is measured using a Millisecond based clock. 500hz = 2ms period, 1khz = 1ms period, the reason it may read some times as 1000hz is the packet may have came in slightly under 2ms from the last control packet that was unpacked, but since its measured in ms the FC would see the time difference as 1 since a full 2ms hasn't elapsed yet. The time between would need to be measured in Microseconds for that to be more accurate which isn't worth the extra overhead right now since that OSD element is mainly to help diagnose any major issues and to give a rough estimate

PGIII, 2022-03-31:
Selectable RSSI Channel Support is now in 10.1.1.5925 |10.1 Alpha 48|

closed 2022-03-31 by PGIII

gretel, 2022-03-31:
> isn't worth the extra overhead right now since that OSD element is mainly to help diagnose any major issues and to give a rough estimate

it's mainly used to show off expresslrs, right :)

> Selectable RSSI Channel Support is now in 10.1.1.5925 |10.1 Alpha 48|

cool thanks.