FlightOne/FalcoX-Issue-Tracker #279: OSD over DisplayPort for SharkByte stopped working with latest SharkByte firmware version 20102021

Tigereye2k6, 2021-10-30:
**Version**
10.1.1.5792

**Target**
TMOTORF7

**Describe the bug**
The OSD stopped working with the latest SharkByte Firmware 20102021. I have flashed the latest FalcoX Firmware but it still wont work. With the previous SharkByte Firmware version all worked well.

The max buad rate is set correct to 115200 and i see sometimes the start message "FALCOX OVER DISPLAY PORT" but this message stays and the osd wont get displayed. Most of the times nothing will be displayed.

Then i have flashed BetaFlight and there the OSD works as expected without any issues :/

PGIII, 2021-10-30:
Sounds like they must have changed something we will have to look into it

Tigereye2k6, 2021-10-30:
A little update: I have downgraded the SharkByte firmware to the previous version and the OSD works. So there is definitely some issue with the new firmware version and falcox.

ligenxxxx, 2021-11-04:
Try to disconnect the TX on the VTX side to see if the OSD can be displayed normally.

Tigereye2k6, 2021-11-04:
I will try it 😉

Tigereye2k6, 2021-11-04:
Have tried it, but it wot work either :/

AzulloCyll, 2021-11-18:
OSD not working on H7 Lightning too! Same SB firmware.

PGIII, 2021-11-30:
Just pushed a new alpha that fixes this issue

closed 2021-11-30 by PGIII

Tigereye2k6, 2021-11-30:
Thank you very much!! ❤️

Tigereye2k6, 2021-12-02:
Have tried it and i can confirm its fixed!