FlightOne/FalcoX-Issue-Tracker #282: Setup Wizard fails to recognize Rx on Racepit

Mikel2332, 2021-11-14:
Flashed a Racepit F4 with FalcoX but Setup fails to recognize the r-xsr (soldered to the designated sBus pad). There's communication since it recognized the vtx and its protocol on the first step.

WildWillyFL1, 2021-11-15:
What UART is the sbus pad?

Mikel2332, 2021-11-15:
I started following the Racepit pinout: Rx3 as to receive Frsky sBus. Failed. Then tried rx1. Failed. I even tried tx1 as if it was a RevoltOSD. Failed too. Afterwards I tried to re-solder the uart3 inverter the other way round (those tiny pads to solder one way or another in order to change the Uart. Sorry I don't know the English word for that). Unfortunately I ended up burning them... I also tried the uninverted sBus pad on the r-xsr on uart1 and uart3. Failed. 

Funny thing is that I ended up reflashing beta fw and didn't recognize the radio either (since the uart3 inverter was burnt I guess). In beta though I succeeded by using rx6 and the uninverted sBus signal from the r-xsr.

Should I reflash FalcoX and see if the uninverted sBus + rx6 works?

Additional note: not sure if the wizard recognized irc Tramp on Uart5 either...

Thanks for the support btw!! :)

WildWillyFL1, 2021-11-16:
Yeah reflash and try the uninverted sbus on RX6 and let me know if that works!

oscarkim007, 2021-11-22:
i try to solder Tx6,Rx6 CRSF Protocol. but it is not working. is it SBUS setup after re-Flash firm? anyone help me!!

WildWillyFL1, 2021-11-24:
No it doesnt have any default protocol, can you try setting the CRSF rx to n.i. sbus and see if it detects?

closed 2021-12-31 by stale[bot]